oro-os/link #16: Add debug pins on signal lines

Qix-, 2023-05-18:
At least for the next few revisions, debug pins should be put on the signal lines for various parts of the board. While soldering on the MCU pin lead wires is kind of fun and looks cool, really slows down the process and introduces a ton of surface area for error. Remove them once the board is ready for a full test.

closed 2023-05-30 by Qix-

Qix-, 2023-06-02:
I've gone back and forth on this about ten times now and decided it's not worth it given they'll only add signal integrity issues.

1. The ethernet SPI lines work, those are already tested.
2. The i2c lines for the indicator lights already work, those are already tested.
3. The SWD lines "work" (ST did not do a _great_ job implementing them) and a programmer is already designed and will be assembled at the same time as the link board. Plus, worst case, I can always solder to the Oro pad as it gives enough space.
4. The only debug lines left are the OLED SPI signals, which are NOT rated for 5v line the arduino pumps out, and I don't have any signal translators on hand (I could get some but.. why) but ARE rated for the MCU's voltages (as far as I can understand the very, very poor documentation). If this doesn't work, I'm going to resort to i2c and just... eat the framerate drop, I guess.
5. All of the power lines and signal lines have THTs associated with them already between the button and the power jack, and the power converter has been tested oodles of times at this point and is known to work.

Other than that, there are no other debug THTs that were marked for population, thus they have all been removed.

weiliddat, 2023-06-07:
> 3. The SWD lines "work" (ST did not do a _great_ job implementing them) and a programmer is already designed and will be assembled at the same time as the link board. Plus, worst case, I can always solder to the Oro pad as it gives enough space.

What’s the difference between using SWD line and using the programmer?

commit df24e5bd (2023-05-30): add debug pins to signal lines (closes #16)